OctoPrint/plugins.octoprint.org #653: [Abandoned Plugin] Touchtest Bed Leveling

rafaelbiasi, 2020-10-08:
**Which Plugin? (URL to plugin page on plugin repository)**

https://github.com/Peaches491/OctoPrint-Touchtest

**Ticket on plugin's issue tracker confirming abandonment:**

https://github.com/Peaches491/OctoPrint-Touchtest/issues/23
https://github.com/Peaches491/OctoPrint-Touchtest/issues/24

**Is there someone open to adopting it? ("no" or @-mention user):**
Unknown

luizmotta, 2020-10-10:
I may be able to help maintaining this plugin but it's my first and I'd have to start from the beginning. Is there a guideline for plugin developers?

cp2004, 2020-10-10:
Hey @luizmotta! There are some good docs that you can use:
* [The plugin tutorial](https://docs.octoprint.org/en/master/plugins/gettingstarted.html)
* [Developing plugins (docs.octoprint.org)](https://docs.octoprint.org/en/master/plugins/index.html)
* [Plugin Development Series (YouTube)](https://www.youtube.com/playlist?list=PL9j2DtsIPVkMsUR457FLodhyDFtKbCkRh)

As well as this, if you get stuck, we're always around in discord/community forums to help out, should you have any questions.

jneilliii, 2020-10-13:
If there is no response we will just update the plugin repo with links to your fork and then submit a notice change. You can see an example of this in the PR [here](https://github.com/OctoPrint/plugins.octoprint.org/pull/651).

Peaches491, 2020-11-05:
@luizmotta and I have resolved the outstanding PRs, and we are picking up continued development. What is needed to mark it as "un-abandoned" ?

cp2004, 2020-11-05:
The abandonment notice can be removed, you can either create a PR to remove the `abandoned:` section of the markdown file or ask nicely and someone will do it for you 🙂

Peaches491, 2020-11-05:
PR Opened: https://github.com/OctoPrint/plugins.octoprint.org/pull/675

jneilliii, 2020-11-06:
@Peaches491 one more thing...need to create the release 0.3 on your repo. https://github.com/Peaches491/OctoPrint-Touchtest/releases

Peaches491, 2020-11-06:
Done! https://github.com/Peaches491/OctoPrint-Touchtest/releases/tag/0.3

closed 2020-11-06 by jneilliii

luizmotta, 2020-11-06:
Great to see the plugin back!