OctoPrint/plugins.octoprint.org #687: [Abandoned Plugin] Filament Sensor Reloaded

ssorgatem, 2020-11-18:
**Which Plugin? (URL to plugin page on plugin repository)**

https://plugins.octoprint.org/plugins/filament_sensor_reloaded/

**Ticket on plugin's issue tracker confirming abandonment:**

https://github.com/ssorgatem/Octoprint-Filament-Reloaded/issues/11

**Is there someone open to adopting it? ("no" or @-mention user):**

Maybe someone from here: https://github.com/OctoPrint/plugins.octoprint.org/issues/499

Rather than silently abandoning the plugin, I'd rather make it known as soon as possible that it needs a new maintainer, and I am willing to transfer the github repo.

AntDX316, 2021-11-03:
why abandon?? just need it to stop when filament is out

AntDX316, 2021-11-09:
how can we add a pause stop command now?

jneilliii, 2022-01-22:
@nickmitchko has volunteered to take over maintenance of the plugin @ssorgatem. If you would transfer the repo to them or provide access that would be the easiest transition.

nickmitchko, 2022-01-24:
Confirming @jneilliii 's statement above

AntDX316, 2022-01-28:
> @nickmitchko has volunteered to take over maintenance of the plugin @ssorgatem. If you would transfer the repo to them or provide access that would be the easiest transition.

does that mean the sensor stops will work again?

ssorgatem, 2022-01-28:
Hi, sorry for not replying earlier, I just saw this.
I'll look into providing access or transferring the repo.

ssorgatem, 2022-01-28:
@nickmitchko I added you as a collaborator, because it wouldn't let me transfer the repo because you have it already forked.

If you delete your fork I can do the transfer.

nickmitchko, 2022-01-28:
@ssorgatem I have deleted the fork, if you prefer the transfer. I am okay with either solution

AntDX316, 2022-01-28:
The sensor stop pause code just has to work when triggered.

cp2004, 2022-01-28:
@AntDX316 please stop posting off-topic comments in this issue, which is about transferring the maintainer of the plugin. Thanks.

PR #1020 (2022-02-28): Filament Reloaded - New Maintainership
Updated the plugin links to reflect new maintainer-ship.

As this is the first time publishing a plugin, please let me know if there are any glaring issues!

Cheers,
nickmitchko

resolves #687
closed 2022-02-28 by jneilliii