OctoPrint/plugins.octoprint.org #698: 404 Detailed Progress

nickm324, 2020-12-01:
When updating Detailed Progress to the recommended version 0.2.6 you get a 404 error.
404 not found https://github.com/dattas/OctoPrint-DetailedProgress/archive/0.2.6.zip

jneilliii, 2020-12-01:
this is the repo for the plugin repository, please post any plugin specific issues in there repository.

jneilliii, 2020-12-01:
The issue I think is because you didn't reinstall from the new maintainer. Try copying/pasting the URL in Plugin Manager > Get More ...from URL and clicking Install.

```
https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/0.2.6.zip
```

closed 2020-12-01 by jneilliii

nickm324, 2020-12-01:
> The issue I think is because you didn't reinstall from the new maintainer. Try copying/pasting the URL below in Plugin Manager > Get More ...from URL and clicking Install.
> 
> ```
> https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/0.2.6.zip
> ```

Well its being done automatically in octoprint, so are you saying I need to uninstall then reinstall? I just installed this less than a week ago and just go a notice today there was an update.

nickm324, 2020-12-01:
> https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/0.2.6.zip

The correct URL worked. Not sure why its pulling the old URL cause I just installed this a week ago brand new...