OctoPrint/plugins.octoprint.org #710: [Abandoned Plugin] Add GCODE like HUE commands to OctoPrint

LMS0815, 2020-12-15:
**OctoPrint-hueCommands**

https://github.com/LMS0815/OctoPrint-hueCommands

**https://github.com/LMS0815/OctoPrint-hueCommands/issues/3**

**Is there someone open to adopting it? NO:**

jneilliii, 2020-12-15:
@LMS0815 are you saying you're abandoning your plugin or just letting us know there is an issue with 1.5.0+ and the plugin so we can blacklist it until you have had time to fix the issue? Would be willing to help out here if I had a Philips Hue set-up, but unfortunately do not.

LMS0815, 2020-12-15:
Blacklisting would be a perfect first step. 
Never the less I wont have much time for coding int the next couple of weeks. 
Therefor I am happy to transfer the project to someone having more time to maintain the code.

I assume the jinja2 Te,plate has a knockout issue which I can not find.

cp2004, 2020-12-15:
I think blacklisting would be a bit *too* far, given that it does not appear to break anything else - usually reserved for plugins that break the server/UI/other functionality/other plugins etc. 

I have a Phillips Hue setup, and was wondering about this. I probably would not adopt the plugin, but maybe I will find time to make a PR. I don't have much time with all the other projects I have on-the-go either, so would rather just send you a fix & you keep the plugin. No guarantees, it gets done **if** I have time.

LMS0815, 2020-12-15:
I really would appreciate some help with the coding.  As the plugin was working well with OctoPrint 1.4 there are probably only small adjustments needed. I just heave no Idea where to search. 

Please use: https://github.com/LMS0815/OctoPrint-hueCommands/tree/develop

LMS0815, 2020-12-16:
I fixed the issue. Thank you.

LMS0815, 2020-12-16:
Updated _devel_ version. Awaiting testing.

closed 2020-12-16 by LMS0815