OctoPrint/plugins.octoprint.org #711: [Abandoned Plugin] <Active Filters>

KenLucke, 2020-12-15:
**Which Plugin? (URL to plugin page on plugin repository)**
Active Filters
https://github.com/MoonshineSG/OctoPrint-ActiveFilters

**Ticket on plugin's issue tracker confirming abandonment:**
Several requests to update to Python 3 (since June) have been unanswered/ignored.  It is now completely  broken due to 1.5.x and Python 3

**Is there someone open to adopting it? ("no" or @-mention user):**

kgleason, 2021-02-09:
I'm going to fork the plugin tonight and see if I can get it working. If I can, then I'll take it.

KenLucke, 2021-02-09:
@jneilliii  has the equivalent plugin which currently works, so you could save some effort if you don't want to duplicate them. I didn't realize that until after I had posted this request.

jneilliii, 2021-02-09:
Mine is a little smarter as well as it stores your settings to config.yaml instead of your browser's local storage, so whatever browser/device you access from the filters are already enabled.

jneilliii, 2021-02-09:
For reference, here's mine...

https://plugins.octoprint.org/plugins/active_filters_extended/

kgleason, 2021-02-09:
Brilliant. That saved me some time tonight. More time for calibrating the printer!

closed 2021-02-23 by KenLucke