OctoPrint/plugins.octoprint.org #719: [Abandoned Plugin] FileManager

evannadeau, 2020-12-18:
The README states the plugin is no longer in active development.

https://github.com/Salandora/OctoPrint-FileManager

https://plugins.octoprint.org/plugins/filemanager/

https://github.com/Salandora/OctoPrint-FileManager/issues/35

I would like to take on the ownership of this plugin, and perhaps get the help of others who have submitted pull requests.

cp2004, 2020-12-18:
If you have a fork/copy of the plugin, then you can submit a PR along the lines of #651 which I did to adopt EEPROM editor.

jneilliii, 2020-12-18:
Let's see if @Salandora can transfer ownership of the plugin repo to @evannadeau. Long-term that's a little easier to handle, but if he hasn't responded to the issue or this comment within 14 days we can do the other process using your own fork.

Sandi2211, 2021-01-11:
What's the status here? I'm really looking for that fix.

jneilliii, 2021-01-11:
you can copy/paste this URL in plugin manager > get more > ...from URL and click Install.

```
https://github.com/jneilliii/OctoPrint-FileManager/archive/fix/jqueryui_deprecation.zip
```

Salandora, 2021-01-24:
Honestly I wouldn't be happy to transfer the ownership of my repo to someone I have no history with and also who doesn't seem to have much interest continuing the development plugin (based on the github profile and fork).

Also @foosel mentioned in https://github.com/OctoPrint/OctoPrint/issues/3596#issuecomment-646537062 that she would like to bundle it into OctoPrint itself, which I would be very happy with.

Also I wonder where the sudden interest in that plugin comes from...

junebug217, 2021-01-24:
I believe the sudden interest in the plug-in is that it has stopped working correctly with the latest rounds of Octoprint updates.

cp2004, 2021-01-24:
I haven't used the plugin, but now that there's move file support in OctoPrint is it just a different UI for the same thing? Maybe it offers some bonus functionality. I've personally never tried it, since I'm not that organised...

I guess a 'file explorer' view could be put into a new UI for OctoPrint at some point in the future - but if you are back (slightly) maintaining this plugin do you want this issue closed? Another option may be to invite a collaborator so you still have ownership, but someone else can handle bug fixes/PRs even if there is no new development. It does have over 6K tracked installs, so there is a fair amount of interest still.

Salandora, 2021-01-24:
hmm "back" is a very good question, I was "back" last year at summer or pre-summer already but then my moving happened and completely throw me off again...
Not sure if I should consider myself "back" at this point in time "again" as I just today looked into github since a few weeks. Maybe let some time pass and see how things turn out.

evannadeau, 2021-01-24:
With Salandora being around, I see no reason to continue with this.  Thank you @Salandora

closed 2021-01-24 by evannadeau