OctoPrint/plugins.octoprint.org #720: [Listing Error] OctoPrint Display Panel

Hairyloon, 2020-12-18:
The link appears to be broken...

**Which Plugin? (URL to plugin page on plugin repository)**

**What's wrong with the plugin *listing*?**

cp2004, 2020-12-18:
'The Link'

You expect us to search through and test each individual link, or maybe tell us?

Either way, please check this is not a duplicate of #717 first.

Hairyloon, 2020-12-18:
The link to the damned installation.
And if you've had it reported and closed the issue, how come you've not fixed it? It's not bleedin' rocket surgery.

cp2004, 2020-12-18:
Excuse me, but that is not how you respond to someone **unpaid** who **volunteers their time** to fix things for the community.

Have a good day.

closed 2020-12-18 by cp2004

Hairyloon, 2020-12-18:
> 
> 
> Excuse me, but that is not how you respond to someone **unpaid** who **volunteers their time** to fix things for the community.
> 
> Have a good day.

You started it friend.
Give a bad attitude and you can expect to get one in reply.

cp2004, 2020-12-18:
Given I thought I had fixed the links 8 hours ago, if you were to be specific then it would really help. Because I did go and have a look after commenting to see if I could find one, and I couldn't find one - so my point still stands.

 If you'd come back and said, "this link right here, `brokenlink.com`" and it was one I thought I had already fixed, maybe there was a mistake. Or if it was the same one, I might have told you how you could refresh the repository to make sure nothing was cached. Come on, it is not much to ask to save me some time and say exactly what the problem is, rather than send me off on a wild goose chase trying to find a random broken link somewhere on the plugin repository.

Hairyloon, 2020-12-18:
Given that the entire purpose of that page is to direct people to the software, "**The** link" clearly refers to the link to the software.
Had it been a random link, then I would have said so.

foosel, 2020-12-18:
@Hairyloon your aggressive outburst here is unacceptable, regardless of the issue at hand. Treat people like you would like to be treated.