OctoPrint/plugins.octoprint.org #729: [Listing Error] Printer Statistics

Flow2105, 2021-01-10:
**https://github.com/AlexVerrico/octoprint-stats/archive/py2.zip**

*when i try to install it it takes a few minutes and then commes this
 Fehler!
Could not install plugin, unknown error, please consult octoprint.log for details
how do i upload the log

cp2004, 2021-01-10:
Please report issues with the plugin itself to the author, linked below for your convenience:

https://github.com/AlexVerrico/octoprint-stats/

There's nothing we can do here to help here, the link works so it is an issue on the plugin's end.

closed 2021-01-10 by cp2004

Flow2105, 2021-01-11:
what can i do to fix this ?

jneilliii, 2021-01-11:
I see you've already opened an issue on the plugin's repository. seems like you'll have to wait for a response from the plugin maintainer.