OctoPrint/plugins.octoprint.org #754: [enforced open drain please flag it properly in dt/acpi dsdt/board file] OctoPrint-Enclosure

firozdang, 2021-01-31:
**Which Plugin? (URL to plugin page on plugin repository)**
https://github.com/vitormhenrique/OctoPrint-Enclosure

**What's wrong with the plugin *listing*?**
I connected DS18B20 sensor with pullup resisters, and after updating /boot/config with dtoverlay=w1-gpio, I cant see anything in dmesg
when I do dmesg | grep gpio, I got the following error: 
enforced open drain please flag it properly in dt/acpi dsdt/board file

jneilliii, 2021-01-31:
This is not the proper place to post bugs with plugins themselves, please use the plugin's repo for that, linked below for your convenience.

https://github.com/vitormhenrique/OctoPrint-Enclosure/issues

closed 2021-01-31 by jneilliii

firozdang, 2021-01-31:
Sorry about that, thanks, will post the issue there!