OctoPrint/plugins.octoprint.org #755: [Listing Error] DisplayLayerProgress

Alainjol, 2021-01-31:
**Which Plugin? (URL to plugin page on plugin repository)**

**What's wrong with the plugin *listing*?**

cp2004, 2021-01-31:
Please don't open blank issues, fill in all the details so that we know what is wrong.

For problems with the **plugin repository itself** then you can file them here, for problems with **anything else** please make sure you are in the right place.

closed 2021-01-31 by cp2004

OllisGit, 2021-01-31:
Hi @Alainjol,

beside the remarks from cp2002: if you want to report an issue about DLP Version 1.25.2, see [issue 205](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/205)  ...and this is fixed in [Version 1.25.3](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.25.3)