OctoPrint/plugins.octoprint.org #758: [Abandoned Plugin] OctoPrint-Mmu2filamentselect

NightSkySK, 2021-02-01:
**Which Plugin? (URL to plugin page on plugin repository)**

https://github.com/derpicknicker1/OctoPrint-Mmu2filamentselect
https://plugins.octoprint.org/plugins/mmu2filamentselect/

**Ticket on plugin's issue tracker confirming abandonment:**

https://github.com/derpicknicker1/OctoPrint-Mmu2filamentselect/issues/28

**Is there someone open to adopting it? ("no" or @-mention user):**
 @tkoecker has prepared the Python 3 compatibility PR and some new features on his fork, maybe he would like to take it over?
https://github.com/tkoecker/OctoPrint-Mmu2filamentselect

tkoecker, 2021-02-03:
I'd be willing to take it over - however to set the expectations right - in the scope of a spare time side project.

jneilliii, 2021-02-03:
Thanks @tkoecker I too do this on my spare time, so that's totally understandable. The only person working full time OctoPrint development is foosel. The only thing we would need from you is a PR to the plugin repo here updating the links to your fork, and your fork needs to have it's software update check point back at your repo and a version bump higher than the current release version in setup.py and github releases. Feel free to reach out if you have any questions.

triantos, 2021-02-05:
I'll volunteer to manage this if @tkoecker does not want to do so, but all the work to port to Python3 appears to be done.

jneilliii, 2021-02-10:
New maintainer now transitioned.

closed 2021-02-10 by jneilliii