OctoPrint/plugins.octoprint.org #763: [Listing Error] SimplyPrint Cloud

dkalinai, 2021-02-02:
The plugin in the subject despite being advertised on the octoprint page is not listed in the repository from Octoprint.

cp2004, 2021-02-02:
It is? https://plugins.octoprint.org/plugins/SimplyPrint/

You may need to wait for the cache on the repository to expire, since it only refreshes every 24 hours.

Here's a screenshot of the button:
[image]

Click that, then come back here and let us know that it is *hopefully* fixed.

dkalinai, 2021-02-02:
Got it, that must be the 24h thing. Oh well, sorry to have bothered everyone. I ended up injecting direct install link from the github repo before this reply arrived.

closed 2021-02-02 by dkalinai