OctoPrint/plugins.octoprint.org #768: [Suspicious Plugin Activity] SimplyPrint Cloud

ShadowOfTheDamn, 2021-02-07:
**Which Plugin? (URL to plugin page on plugin repository)**
https://plugins.octoprint.org/plugins/SimplyPrint/
**What's the suspicious activity?**
after uninstalling the plugin even after restarting the raspberry there is still notation on the LCD about the simplyprint plugin.
**Link to ticket on the plugin's issue tracker about the issue**

jneilliii, 2021-02-07:
@AlbertMN could you please chime in here as to what might be going on? My suspicion is that the GCODE Scripts that are added by SimplyPrint are not getting reset to what they were before and as a result are still sending M117 messages to the printer.

@ShadowOfTheDamn if you look inside your OctoPrint's settings in GCODE Scripts section do you see the message that you see on your printer in any of those sections?

cp2004, 2021-02-07:
It is always better to get in touch with the plugin author first, rather than reporting issues here. It's likely a bug in the plugin, this is not 'Suspicious' really since we have good relationships with the SimplyPrint guys, and they work quickly to get reported issues fixded. They even have an online chat, it's not hard to get in contact with them.

ShadowOfTheDamn, 2021-02-07:
> @AlbertMN could you please chime in here as to what might be going on? My suspicion is that the GCODE Scripts that are added by SimplyPrint are not getting reset to what they were before and as a [...]

I have checked my Gcode Script section of octo print no M117 Command nor any thing regarding that message "[SimplyPrint] DT9"

> It is always better to get in touch with the plugin author first, rather than reporting issues here. It's likely a bug in the plugin, this is not 'Suspicious' really since we have good relationships [...]

Yes but installing and uninstalling the software is before basics. and if the software still tries to connect you to their servers or anything like that I am sorry But I consider it hijacking my HW.

ShadowOfTheDamn, 2021-02-07:
The solution to the problem is to uninstall the plugin without selecting to delete the data. if you select to remove the data you will get this behavior. Literally the developers could try uninstalling the software once?
Sorry for everybody's time.
case dismissed.

closed 2021-02-07 by ShadowOfTheDamn

cp2004, 2021-02-07:
I'm sure they didn't do it deliberately, it's always better to ask nicely first.

AlbertMN, 2021-02-07:
Hi @ShadowOfTheDamn :) Very sorry you had this issue, and of course it isn't intentional, and of course we tried to uninstall our software (too many times to count).

We've had this report once, and we're looking into it and coming fix a fix very soon. I can't stress this enough; it's of course not the expected behaviour, and we apologise for the inconvenience! We are currently in Beta, just rolled out internationally, so it's been a bit stressful, and we haven't been able to get to this issue just yet, but it's first on my list for tomorrow :D

We're super fast to respond to any emails we get, and we have a live chat on our website, which is open 8 hours a day _(on week days only, though)_, so you're very welcome to direct any and all issues and criticism our way directly; we sure deserve this one, and don't take it lightly.

Will write a comment on this issue when the problem is tested & fixed - thanks for bringing this to our attention, and I do get why you find it suspicious and a"hijacking". This is of course not what we were aiming for, and hope you understand the chaos it can be to release such a huge system as this :)

Thanks for the tag @jneilliii

ShadowOfTheDamn, 2021-02-08:
@AlbertMN thanks for the reply. I get what you say, no worries. good luck.