OctoPrint/plugins.octoprint.org #770: [Abandoned Plugin] OctoPrint EEprom Repetier Editor

awenelo, 2021-02-08:
**Which Plugin? (URL to plugin page on plugin repository)**

https://plugins.octoprint.org/plugins/eeprom_repetier/
(https://github.com/Salandora/OctoPrint-EEprom-Repetier)

**Ticket on plugin's issue tracker confirming abandonment:**
In README.md, it says: `Plugin actually not in active development!`. Additionally, a PR to add Python 3 compatibility in October (Salandora/OctoPrint-EEPROM-Repetier#12) has received no response.

**Is there someone open to adopting it? ("no" or @-mention user):**
@Wotsamatter has a branch with Python 3 support and some other changes, maybe they would be willing to take it over?

jneilliii, 2021-02-08:
I know @Salandora is still around and occasionally threatens of active development again 😄, but life happens. Maybe he would be willing to transfer ownership of the repo, or potentially add someone else that is willing to take over maintenance? Once we have a volunteer to do so.

Salandora, 2021-02-14:
Well in this case here I would be happy to add some maintainers, I don't use Repetier anymore so I wouldn't be the best person to continue the plugin as I have no real way to test it.
So if there are any volunteers please feel free to post below this thread.

Wotsamatter, 2021-02-16:
Not a big audience for this plugin I expect, but I have found it very useful for tuning the settings on my Repetier-based printer (thanks Salandora). I can maintain the existing codebase going forward to Python 3.x to support any other users out there and assist the general migration effort to Python 3, so count me as a volunteer.

jneilliii, 2021-02-16:
That's great news @Wotsamatter, and thank you @Salandora for offering to add a maintainer. That will make the process a lot easier to transition. Just let us know here once this has been completed and we can close this issue, etc.

Salandora, 2021-02-26:
@Wotsamatter you should have gotten an invitation to become a collaborator on the repo right now.

jneilliii, 2021-02-26:
Thanks @Salandora now that @Wotsamatter should have access I'll go ahead and close this out.

closed 2021-02-26 by jneilliii

Wotsamatter, 2021-02-28:
> <plugins.octoprint.org@noreply.github.com>
> <mention@noreply.github.com>
> OctoPrint EEprom Repetier Editor (#770)

Hi,

Thanks, I have just accepted.

Regards from Melbourne,
Lloyd.