OctoPrint/plugins.octoprint.org #785: Rant: Plugins that modify gcode files are a plague

eyal0, 2021-02-19:
I don't know if anything can be done about it but I would like to at least draw attention to the problem: Plugins that work by modifying gcode are a plague on the OctoPrint plugin repo.

I get bug reports on my plugins with people complaining that their combination of plugins isn't working well and I suspect that the cause is because there are plugins that perform their actions by modifying gcode.  Things like:

* A plugin that wants to display info on the display might fill a file with `M117` commands.
* A plugin that wants to optimize gcode might use the [Douglas–Peucker algorithm](https://en.wikipedia.org/wiki/Ramer%E2%80%93Douglas%E2%80%93Peucker_algorithm) to decrease file size.

The problem is when all these plugins try to read and write gcode files simultaneously.  If you're lucky, they'll work one at a time.  Usually, however, they will trigger one another leading to lots of duplicated work.

There's a better way: OctoPrint has hooks for filtering gcode before it goes to the printer.  Store metadata for the file that needs modification indicating the changes.  That metadata is a sort of "diff" on the original gcode.  Then apply that diff on-the-fly with a gcode hook.

jneilliii, 2021-02-19:
I'm not sure this is the correct place to have this discussion, but we have been talking about this very thing on the Discord channel and creating a configurable priority for these types of modifying hooks was the specific use case. @kantlivelong was working on a pull request the last I saw after cleaning up the code a bit. Maybe that's something that would assist in your situation.

kantlivelong, 2021-02-19:
If using hooks with well thought out implementation there shouldn't be and kind of accessing gco files at the same time. It's when plugins try to do this using events that this becomes an issue. There is also **some** responsibility on the user to know what a plugin is doing and they should be isolating the issue when reporting a bug. If two plugins are indeed conflicting with each other then the user can certainly point it out in a report but as a maintainer you're not obligated to fix it.

> There's a better way: OctoPrint has hooks for filtering gcode before it goes to the printer. Store metadata for the file that needs modification indicating the changes. That metadata is a sort of "diff" on the original gcode. Then apply that diff on-the-fly with a gcode hook.  

Doing stuff on the fly is usually a good way to go about it if it's a quick operation but for things like ArcWelder it's not realistic.

> @kantlivelong was working on a pull request the last I saw after cleaning up the code a bit. Maybe that's something that would assist in your situation.

I am indeed working on a change that will bring some additional flexibility with hook ordering. It will allow plugins to define order that is _after_ hooks with no defined order. Also will allow users to adjust the ordering. This won't really help though if the plugins implementation isn't sound to begin with.

PS: This also really belongs on the forum/discord and not a bug tracker (especially this one).

closed 2021-02-19 by jneilliii