OctoPrint/plugins.octoprint.org #797: [Abandoned Plugin] OctoPrint-Slack

awenelo, 2021-02-24:
**Which Plugin?**

https://plugins.octoprint.org/plugins/slack/

**Link to ticket on plugin's issue tracker confirming abandonment**

https://github.com/richjoyce/OctoPrint-Slack/issues/17

**Is there someone open to adopting it?**

Don't think so

mkevenaar, 2021-02-26:
I am willing to try to maintain this. How do I get started?

jneilliii, 2021-02-26:
The first step would be creating a fork of the plugin and update all its software update hooks and such to point to your repo on github. Then make whatever necessary changes are there for python 3 compatibility and release a new version updating the setup.py file with the new version. Once that is done, submit a PR against this repo changing the plugin's listing file to point to your github repo, add the authors: list to include yourself. We can help you through these last steps as part of our review of your PR.

mkevenaar, 2021-02-26:
I can do that, will be working on that tomorrow! Thanks

closed 2021-02-28 by jneilliii