OctoPrint/plugins.octoprint.org #807: [Abandoned Plugin] WebcamStreamer

jneilliii, 2021-03-04:
### Which Plugin?

https://plugins.octoprint.org/plugins/webcamstreamer/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/adilinden-oss/octoprint-webcamstreamer/issues/35#issue-706491348

jneilliii, 2021-03-04:
Yeah, I saw that one too. Working on the PR now to flag.

foosel, 2021-03-04:
I was just about to ask, will you or should I ;)

jneilliii, 2022-01-25:
Now that my RTMPStreamer plugin works the same as this plugin with the added benefit of an optional docker requirement (can stream using ffmpeg directly) maybe this plugin could be redirected to mine @foosel or remove the listing altogether?

foosel, 2022-01-26:
We can adjust the description to point to yours, but an automatic redirect feels a bit like a "hostile takeover" given that it is indeed a different plugin from a different author. 

We might have to think about a general approach on how to handle plugins that have been abandoned without any obvious interest to adopt them for a long duration. But that's probably better discussed on Discord.

adilinden, 2022-02-03:
I am not going to be able to look at https://github.com/adilinden-oss/octoprint-webcamstreamer any time soon. Whatever you think is best to efficiently get people searching for "octopi google streaming" to a functioning plugin I am ok with. 

I only noticed this because yesterday I turned on my printer after a long hibernation (lack of time and space to pursue) and thought too look if there have been advances by others that  would my work obsolete.

closed 2022-04-13 by jneilliii