OctoPrint/plugins.octoprint.org #816: [Listing Error] Octoprint Youtube Timelapse

3DMakeShack, 2021-03-09:
### Which Plugin?

https://plugins.octoprint.org/plugins/youtube_timelapse/

### What kind of listing error are you reporting?

broken link on the listing

Nothing happens.  Nothing installs.

3DMakeShack, 2021-03-09:
Plugin doesn't shop up in the Add more library.  I installed using the installation url provided, but nothing happens.  NOTHING happens.  Nothing Installs.  I did my diligence at Google with the instructions that no longer match what Google displays, but I was able to get through it only to have the plugin not install.

cp2004, 2021-03-09:
Please report issues with specific plugins in the bug tracker of the plugin instead of here. This tracker is for reporting issues with the plugin repository, not issues with the plugins registered in it, as also stated on the ticket form:

> What kinds of issues you should report here:
> 
> * missing or broken homepage, source or download URLs
> * broken images
> * incorrect compatibility information
> * ... in short, anything related to the data on plugins.octoprint.org
> 
> What you should not report here:
> 
> * Any kinds of issues with the plugins themselves - report those to the plugin author instead!

The reason it doesn't show up in the 'Add more' library is that it is either incompatible to your system, or the cache on the repository has not expires (see #763 for a screenshot of the 'refresh' button)

Also, the 'nothing' happening is very difficult to work out without a log or screenshot of what is going on, please share this next time.

Please seek support on the [community forums](https://community.octoprint.org) or with the plugin's author - I have verified the download URL works by clicking https://github.com/ryanfox1985/OctoPrint-Youtube-Timelapse/archive/master.zip

closed 2021-03-09 by cp2004

jneilliii, 2021-03-09:
Going to make a wild guess here that you're trying to install in python 2 environment as the plugin is python 3 only.