OctoPrint/plugins.octoprint.org #822: OctoKlipper plugin looking for new maintainer

AliceGrey, 2021-03-13:
I would like to pass the project off to another interested developer. If you would like to maintain this plugin please respond here. I know there are a couple of you who have made multiple PR's recently who might be interested. 

P.S. if this isn't the right place for this discussion let me know. I figured this would be similar to an abandoned plugin issue.

AliceGrey, 2021-03-13:
@thelastWallE

AliceGrey, 2021-03-13:
@ollyfg

thelastWallE, 2021-03-13:
Would like to.
Thanks for the consideration. I have a fork that i would like to merge in. But i think i will get the release candidates channel system in before some new features.

AliceGrey, 2021-03-13:
@thelastWallE Go ahead and open a PR on the octoprint plugin repo here to change the OctoKlipper plugin to direct to your repo. You can tag me in it and I'll approve it so the octoprint devs know it's ok.

cp2004, 2021-03-13:
@AliceGrey it would make life easier for you to transfer the repository @thelastWallE - since then Github sets up all of the redirects, so things like software updates still work.

A PR for adoption can still be done to move it to the new location after, but it would not need the notices since everything is handled seamlessly.

AliceGrey, 2021-03-13:
> @AliceGrey it would make life easier for you to transfer the repository @thelastWallE - since then Github sets up all of the redirects, so things like software updates still work.
>
> A PR for adoption [...]

That's a good idea. @thelastWallE I'll go ahead and transfer over the repo to you.

AliceGrey, 2021-03-13:
@thelastWallE Looks like I can't transfer since you already have a fork on Github.

thelastWallE, 2021-03-13:
i renamed it. Hopefully this is sufficient.

AliceGrey, 2021-03-13:
> i renamed it. Hopefully this is sufficient.

Still won't let me since both repo's are forks of the same repo.

thelastWallE, 2021-03-13:
Ok i duplicated my fork into a new repo and deleted the fork afterwards. Should work now.

AliceGrey, 2021-03-13:
> Ok i duplicated my fork into a new repo and deleted the fork afterwards. Should work now. 

Request sent

thelastWallE, 2021-03-13:
All done on my side. We can close this for now.

cp2004, 2021-03-13:
👍

closed 2021-03-13 by cp2004