OctoPrint/plugins.octoprint.org #833: [Abandoned Plugin] Action Trigger Plugin

Booli, 2021-03-31:
### Which Plugin?

https://plugins.octoprint.org/plugins/actiontrigger/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/Booli/OctoPrint-ActionTriggerPlugin/issues/6

jneilliii, 2021-03-31:
Hey @Booli I'd be willing to maintain this plugin for you. It's pretty straight forward and I don't see much in the line of issues/support requests. It would be easiest to just add me as a maintainer to the repo I think and we can then just keep it all on your github, unless you want to transfer the repo to me.

Booli, 2021-03-31:
That is awesome @jneilliii 👍 I've invited you as colaborator

jneilliii, 2021-03-31:
invite accepted, closing issue and reverting abandonment flag.

closed 2021-03-31 by jneilliii