OctoPrint/plugins.octoprint.org #843: Adoption of CR-10 Leveling plugin

electr0sheep, 2021-04-15:
[CR-10 Leveling Plugin](https://github.com/electr0sheep/OctoPrint-Cr10_leveling)

I originally created this plugin years ago, and don't really have time to maintain it any more.

I am looking for someone who wants to take ownership of this plugin. If you are interested, please leave a comment here.

Thanks!

lsellens, 2021-04-18:
If no one else is interested I can take over. I use this plugin all of the time so I don't see myself abandoning it anytime soon.

jneilliii, 2021-04-18:
Thanks @lsellens. The easiest way to transition this is to either have @electr0sheep transfer the repo to your github account, or add you as a maintainer to his repo. Once that's done I can remove the adandoned flag.

AvanOsch, 2021-04-18:
I basically feel the same as @lsellens.
@electr0sheep, maybe you can transfer to one of us, and make the other a maintainer?

icancwhatusay, 2021-04-20:
Well sadly I don't know enough about how to maintain such things but I have found this plug-in to be wonderfully simple, easy to use and incredibly useful so thank you for creating it in the first place.

electr0sheep, 2021-04-23:
@lsellens If you will delete the fork you have, I will transfer ownership to you. It won't currently allow me to do it because you already have a repo with the same name.

lsellens, 2021-04-23:
@electr0sheep done

electr0sheep, 2021-04-23:
@lsellens Transfer has been initiated. I think you just need to "accept" the transfer now or something

lsellens, 2021-04-23:
Cool thanks, Ill keep it alive. @AvanOsch I sent you an invite to be added as a maintainer.

lsellens, 2021-04-23:
@jneilliii you can close this and remove the abandoned flag

closed 2021-04-23 by electr0sheep

jneilliii, 2021-04-23:
@lsellens make sure to adjust the url's in the repo to your github account. The repo transfer will handle software update stuff for future versions, but would be good to have that in place for new installs. Thanks for the smooth transition @electr0sheep.