OctoPrint/plugins.octoprint.org #848: [Listing Error] Creality 2x temperature reporting fix

blaekipups, 2021-04-20:
### Which Plugin?

https://plugins.octoprint.org/plugins/ender3v2tempfix/

### What kind of listing error are you reporting?

incorrect compatibility information on the plugin listing

### What specifically is broken or wrong?

Hey, 
I have a temperature display issue with my Ender 3V2. This Plugin did fix the problem when I first installed it but now when I look at the console, i see the error again. I tried to reinstall the plugin, but it did not fix the problem. 
Could some of my other plugins that control the temperature (Navbar Temperature Plugin (0.14); Preheat Button (0.7.2); Smart Preheat (0.0.6)) cause this Problem? I tried deactivating them, but the error is still there.
I am very new to 3d printing etc, so please try to help me in simple language. Thank you

cp2004, 2021-04-20:
Please report issues with specific plugins in the bug tracker of the plugin instead of here. This tracker is for reporting issues with the plugin repository, not issues with the plugins registered in it, as also stated on the ticket form:

> What kinds of issues you should report here:
> 
> * missing or broken homepage, source or download URLs
> * broken images
> * incorrect compatibility information
> * ... in short, anything related to the data on plugins.octoprint.org
> 
> What you should not report here:
> 
> * Any kinds of issues with the plugins themselves - report those to the plugin author instead!

Thanks :)

In this specific case, the plugin is performing as expected - it cannot do the serial communication for the printer, it just allows OctoPrint to read it.

closed 2021-04-20 by cp2004

blaekipups, 2021-04-20:
Thank you for your Help, next time I will report it in the right place.