OctoPrint/plugins.octoprint.org #860: [Feature Request] Badge/Shield or API for 'Active instances the past month'

Sebclem, 2021-04-26:
### What do you want to see implemented in the plugin repository?

It can be interesting to have the ability to get a badge for `Active instances the past month` so that we can add it to our Github Readme.

Or maybe just have access to an API so we can create one with shield.io

### Why, what problems would be solved by this?

I'm just thinking about adding a Badge/Shield to my github readme

jneilliii, 2021-04-26:
So there is a way to do shield.io badges from your github repo directly for download counts, but requires a little work on your releases. Instead of using the default release you have to use a release asset, which requires a change to the software update check in most cases. There are a couple of plugin authors that are doing this, like https://github.com/cp2004/OctoPrint-WS281x_LED_Status or https://github.com/OllisGit/OctoPrint-DisplayLayerProgress.

cp2004, 2021-04-26:
Better example for the adjusted SW update hook is my WLED plugin rather than WS281x, where the work is only half done.

Sebclem, 2021-04-26:
I find (thanks to @cp2004 's OctoPluginStats) that there is a kind of API with https://data.octoprint.org/.

So we can use https://data.octoprint.org/export/plugin_stats_30d.json with [Shields.oi dynamic badge](https://www.shields.io/#dynamic-badge)

closed 2021-04-26 by Sebclem

foosel, 2021-04-26:
As an example:

```
https://img.shields.io/badge/dynamic/json?label=Instances&query=%24.plugins.octolapse.instances&url=https%3A%2F%2Fdata.octoprint.org%2Fexport%2Fplugin_stats_7d.json
```

[image]

cp2004, 2021-04-26:
I was just about to ask if there was an example of how that would work!

@Sebclem If you want to fork and run your own OctoPluginStats, I'll fix the instructions of 'how to do so'... It's new, and done in Material UI & React at the moment.

OllisGit, 2021-05-09:
fyi: In my KnowHow-Wiki is a description how I made the badges:
https://github.com/OllisGit/OctoPrint-KnowledgeBase/wiki/DownloadCounter