OctoPrint/plugins.octoprint.org #882: [Listing Error] WiFi Status

apbarratt, 2021-05-15:
### Which Plugin?

https://plugins.octoprint.org/plugins/wifistatus/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

It seems the installation url is incorrect, it goes to https://github.com/ManuelMcLure/OctoPrint-WiFiStatus/archive/master.zip which results in a 404.

Looking at the linked homepage, there is no branch named master.  There is however a branch named main.

I'm not sure whether this is a mistake that has taken place on the Octoprint listing side or the plugin's git repo.  I'll create an issue on the plugin repo too and link back to this issue here.

apbarratt, 2021-05-15:
I've gone ahead and created the issue now on the plugin's own repo, hopefully some conversation can be done to figure out which side needs fixed :) https://github.com/ManuelMcLure/OctoPrint-WiFiStatus/issues/18

cp2004, 2021-05-15:
For once, an *actual* listing error. _Actually_ posted in the right place. Not some other bug or someone else's problem...

This does need adjusting in this repository, so we can either wait for @ManuelMcLure to send a PR or someone will adjust it soon.

apbarratt, 2021-05-15:
Aha, do I get a prize? XD

jneilliii, 2021-05-15:
🍪

PR #884 (2021-05-15): wifistatus bugfix
fix install URL, resolves #882
closed 2021-05-15 by kantlivelong