OctoPrint/plugins.octoprint.org #897: [Listing Error] OctoPrint-Pushjet

sir106, 2021-06-10:
### Which Plugin?

https://plugins.octoprint.org/plugins/pushjet/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

service PushJet seems to be down; linked URL (https://pushjet.io/) no longer available.

cp2004, 2021-06-10:
Plugin appears abandoned, and the service is indeed down - I found https://github.com/Pushjet, but that is also inactive.

If we don't hear back from @dipusone my vote would be that this plugin is removed from the repository - if the service is down, then there is not much point in having a plugin listed that can interact with it.

foosel, 2021-06-10:
Another vote for that from my side, for the same reason.

jneilliii, 2021-06-10:
I agree, if it's dead the plugin should be removed. Will wait 7 days to give @dipusone a chance to reply.

closed 2021-06-22 by foosel