OctoPrint/plugins.octoprint.org #900: [Feature Request] Add sort by number of installations

gobbledygoober, 2021-07-04:
### What do you want to see implemented in the plugin repository?

Suggest that "Sort By" active or total installations be added to complement Name, Date, Tag, and Author.

### Why, what problems would be solved by this?

This would be helpful to show new users which plugins are popular. Instead, I looked for articles on the internet that recommended plugins and looked at the raw json export "instances" statistics.

cp2004, 2021-07-04:
If I remember right there is already a sort option in the plugin manager actually inside OctoPrint, just not on the website.

jneilliii, 2021-07-04:
Yeah, I think the default sort order in get more of plugin manager was changed to sort by number of installs.

gobbledygoober, 2021-07-06:
Aha! I see it now. Thanks!

closed 2021-07-06 by gobbledygoober