OctoPrint/plugins.octoprint.org #911: [Abandoned Plugin] Full-featured Slicer

ncot-tech, 2021-07-20:
### Which Plugin?

https://plugins.octoprint.org/plugins/slicer/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/kennethjiang/OctoPrint-Slicer

ncot-tech, 2021-07-20:
From the top of the README.md...

Note: This project is no longer maintained. The latest Cura versions have excellent support to directly slice and send gcode to OctoPrint, and hence render this plugin largely obsolete.

If you are interested in taking over the ownership of this plugin, please PM me.

cp2004, 2021-07-20:
#572 was created previously, so there is someone with ownership but I guess the note on the readme has not been changed?

foosel, 2021-07-20:
Ah, that explains why this sounded so familiar. Thanks for digging up that ticket, @cp2004. 

@eyal0, can you elaborate on what's the status here?

jneilliii, 2022-04-27:
@eyal0 we haven't heard back from you on this and have received another abandoned plugin report. Are you still maintaining the plugin or should we put it up for adoption?

eyal0, 2022-04-28:
I don't have time to maintain this anymore.  I'm willing to help out whoever follows.

urobot2011, 2022-06-29:
Me

jneilliii, 2022-06-29:
@eyal0 would you be willing to transfer the repo to volunteer @urobot2011? We've found that this is the smoothest way to transition a plugin between maintainers.

eyal0, 2022-06-30:
Sure!  How do I do it?

jneilliii, 2022-06-30:
You go into the settings of the repository and use the transfer button at the bottom of the page.

[image]

But before you do that I'd like to get a formal response from @urobot2011 that he truly wants to take over maintenance of this plugin and verify it's not just some random bot on github replying to issues with single word replies.

urobot2011, 2022-08-21:
i not bot.

realbrianosaurus, 2022-09-28:
I like this plugin. I haven't been a programmer for a little while, and I'm not totally familiar with the specifics of using github. How much work is it to maintain the plugin?

jneilliii, 2022-09-28:
It largely depends on the popularity of the plugin and how many people are using it. Dealing with feature requests and bug reports, etc. Currently the plugin is showing 21 open issues. 

https://github.com/kennethjiang/OctoPrint-Slicer/issues

TigerTy9, 2023-02-13:
I can maybe take this on!

jneilliii, 2023-02-14:
Python and JavaScript are both involved with plugins.

jneilliii, 2023-02-14:
BTW the [GridSpace ](https://plugins.octoprint.org/plugins/gridspace/) plugin should be able to work in a mobile browser...

TigerTy9, 2023-02-14:
@jneilliii would you just recommend me using that rather than up keeping this plugin per-say?

jneilliii, 2023-02-14:
I'm not saying that at all, just letting you know there are options.

Garr-R, 2023-07-20:
Hello! I would love to take on maintaining the OctoPrint-Slicer! 

I've slowly been chipping away at combining the former [OctoPrint-Slicer ](https://github.com/kennethjiang/OctoPrint-Slicer)and [OctoPrint-Slic3r ](https://github.com/OctoPrint/OctoPrint-Slic3r)plugins in order to provide native PrusaSlicer support. This allows people to use their own PrusaSlicer printer profile within the very nice OctoPrint-Slicer web interface:

> https://github.com/Garr-R/OctoPrint-Slicer

Currently the plugin works nicely and allows users to download PrusaSlicer v2.4.2 onto their Raspberry Pi from the plugin's settings menu. If approved, should I backup and then delete my own repository? Or should I just rename it so ownership of the original repo can be transferred...?

jneilliii, 2023-07-20:
There's a couple of approaches here @Garr-R and it will depend on how @eyal0 or @kennethjiang want to proceed. If they are willing to transfer the repo to you, then yes you'll need to delete your fork. Otherwise you can keep your fork open and we just point everything to you and setup update check overlays, etc.

eyal0, 2023-07-20:
OctoPrint-Slic3r is just for running slic3r or the prusaslicer.  OctoPrint-Slicer is the GUI interface that can arrange elements on the surface and then sent them to a slicer to print.

Which one did you do, @Garr-R ?

Garr-R, 2023-07-21:
I've basically merged both of them @eyal0. So now the GUI from the OctoPrint-Slicer plugin is integrated with the backend of the OctoPrint-Slic3r plugin, supporting both slic3r and PrusaSlicer profiles and eliminates the need for both plugins to be installed.

I've also added the ability to download the PrusaSlicer to the user's Raspberry Pi from the plugin's settings menu, making the [installation process for new users much, much easier](https://github.com/Garr-R/OctoPrint-Slicer/wiki/Downloading-PrusaSlicer-to-OctoPrint) (assuming they use the default "pi" username...). There is also an offline branch for users that don't have their Raspberry Pi connected to their network (e.g. Colleges with strict network policies that don't allow Raspberry Pi's to connect to their network)

My next big plan for this plugin (outside of getting PrusaSlicer v2.5+ to work on a Raspberry Pi....) is to re-design how profiles are selected and imported. I'd like to build an OctoPrint Wizard that lets users select from the official "vendor" profiles that PrusaSlicer uses (print, filament, and printer presets), which should make it much easier to switch between different filament, nozzle size, and layer height/speed presets. 

Ultimately, I really wanted to get this plugin back up and running for the school districts that use my company's printers (MakerGear). There are a LOT of middle/highschoolers that continue to use this plugin because their Chromebooks can't install PrusaSlicer/Cura, so OctoPrint-Slicer really makes things easier for them. This plugin is also great for all of the people who are getting into 3D-printing, or are slicing on mobile devices with their own profiles.

I'm very open to and looking for feedback! My current release is fully functional and should be a fairly smooth setup process if anyone wants to try it out

If users request support for CuraEngine I'd be open to adding it back into the plugin! ([possibly integrating it with the CLI wrapper from this repo](https://github.com/Spiritdude/Cura-CLI-Wrapper)) I'm just way more familiar with the slic3r/PrusaSlicer's CLI so I started there

eyal0, 2023-07-21:
Is it not better to keep them separate?  You might have users that want just the slic3r/prusaslicer.  Or you might have users that want the 3d interface but without slic3r, because they want to use, for example, cura slicer.

Further still, you might split the slic3r plugin into one for slic3r and one for prusaslicer.  They are sufficiently different now, right?

Finally, and this is only if you're really devoted, I made my own version of OctoPrint-Slicer that has some new features, like the ability to cut an object into two pieces, for example.  And it has an auto-arrange feature that will arrange multiple items on the build plate somewhat optimally.  Also, it can detect objects that overlap or objects that are outside the bounds of the printer.  Anyway, if you're interested in that stuff, I maintained the plugin here: https://github.com/eyal0/OctoPrint-Slicer  I don't promise that it works well.

Anyway, stuff to think about.