OctoPrint/plugins.octoprint.org #917: [Abandoned Plugin] HeaterTimeout

awenelo, 2021-07-29:
### Which Plugin?

https://plugins.octoprint.org/plugins/heatertimeout/

### Link to ticket on plugin's issue tracker confirming abandonment

Github repository is archived, a ticket can't be created

strfr, 2021-07-30:
What is required for maintaining this plugin?

cp2004, 2021-07-30:
Fork the repository, make some changes & adjust the software update hook to point to your fork. Please retain credit to the original author.

Once you have a repository you are using to maintain the plugin, you can submit a PR here adjusting the URLs and adding yourself as an author to the plugin listing.

Joedmin, 2021-07-31:
I may have adopt it, I'll look at it and try to resolve some issues. I wanted to try to make plugins anyway

cereyanli, 2021-09-23:
@Joedmin have you made up your mind ? are you going to take over the development ?

Andy-ch, 2021-10-18:
If the seat is still vacant, I could take it. My Octoprint UI lead me to this page
I've been working with Python for a long time, and the plugin repo is 68% Python

Andy-ch, 2021-10-18:
Bit of a legal question: can I remove [CONTRIBUTING.md](https://github.com/Andy-ch/OctoPrint-HeaterTimeout/blob/master/CONTRIBUTING.md) file? As it is, it requires contributors to sign the [Google Individual Contributor License Agreement](https://developers.google.com/open-source/cla/individual?csw=1), and with the plugin not being maintained by Google, I don't think it's relevant anymore. But a confirmation would be nice

jneilliii, 2021-10-19:
I'm assuming since you forked it from google that is a default they put on all their repos I bet. It seems that @precision was the primary contributor to the actual repo itself so maybe they could provide some insight into that. The plugin is Apache 2 license, so you technically would only have to follow the required limitations of that license.

taylorturner, 2021-10-19:
I'm happy to help out with the Python on this plugin or any other! Would love to contribute to the Octoprint project since I use it so much!

closed 2021-10-20 by jneilliii