OctoPrint/plugins.octoprint.org #918: [Abandoned Plugin] LEDStripControl

awenelo, 2021-07-29:
### Which Plugin?

https://plugins.octoprint.org/plugins/ledstripcontrol/

### Link to ticket on plugin's issue tracker confirming abandonment

Github repository is archived, a ticket can't be created

Jordan-Morales, 2021-08-18:
What all is involved in caring for a plugin. 👀

jneilliii, 2021-08-18:
Mainly just dealing with support related issues, and if needed bug fixes/feature requests at the discretion of the adopter. If you are interested in taking over this plugin there is a small process that has to be completed to transfer the pointers in the plugin repo to your fork of the plugin.

LunkwillAndFook, 2021-10-20:
Give me a few days to look at the code. I may take this since I use the plugin.

dzikk, 2022-01-15:
Hi
I want to use this plugin only with white led, How can I modify the source code to make possible to toogle state of led with only one command? currently I have to use M150 W and M150

OllisGit, 2022-01-15:
Hi @dzikk,
this is not the right place to ask this question, because this repository is only for managing the plugins-page: https://plugins.octoprint.org. 
The plugins itself are maintained in there own repository....but this LED-Plugin is archived..no change to raise the question there.

You should use the community forum: https://community.octoprint.org/
There is a much bigger audience for answering your questions.

Regarding your question: I looked into the sourcecode and from my perspective you should implemente something like this (pseudo-code):

New command type: M150 TW:  T = toggle
```
if cmd == 150
  if cmd.contains(T)
      onOrOff = readGPI from gcode seleced LED (e.g. W or R or B)
      if onOrOff == true
        dutycycles = {'w':0.0}
        self._leds[l].ChangeDutyCycle(dutycycles[l])
  else
    ...do the old stuff
```

gbonesso, 2022-09-26:
Hi! I'm interested in adopting this plugin. This plugin and it's not much complicated and it's a good place to start contributing to the community. What should I do to start? I've already forked it here (https://github.com/gbonesso/OctoPrint-LEDStripControl)...

jneilliii, 2022-09-27:
Modify all the files in your fork to reference it's GitHub address instead of your own, including setup.py and update hook in init.py and then submit a PR here modifying the [ledstripcontrol.md](https://github.com/OctoPrint/plugins.octoprint.org/blob/gh-pages/_plugins/ledstripcontrol.md) template file adjusting the URLs and switching author to a list called authors and adding yourself to it, like this.

```
authors:
- gbonesso (or however you want your name to display)
- Uriah Welcome
```
You'll want to make sure you have a version released on your repo that at least matches the current version release, so updating the version in setup.py to 0.3.8 and creating a release with all the above changes in place.

jneilliii, 2022-09-27:
@gbonesso here's a pretty good example of another plugin adoption and what all files need to be updated for the plugin repository PR mentioned above. 

https://github.com/OctoPrint/plugins.octoprint.org/pull/1027/files

gbonesso, 2022-09-27:
@jneilliii , I've made the changes and submit the PR (https://github.com/OctoPrint/plugins.octoprint.org/pull/1107). I also reinstalled the plugin using my github repository ( https://github.com/gbonesso/OctoPrint-LEDStripControl/archive/master.zip) and it worked fine...

gbonesso, 2022-09-27:
@jneilliii One update: I realized that the screenshot in the plugin page was not updated and I've made a new PR (https://github.com/OctoPrint/plugins.octoprint.org/pull/1108) with the correct screenshot.

PR #1110 (2022-10-02): updates related to adoption of ledstripcontrol plugin
@gbonesso please change your release on your GitHub repo to not be pre-release. Once that is done I can merge these changes and the adoption process will be complete. resolves #918
closed 2022-10-02 by jneilliii