OctoPrint/plugins.octoprint.org #955: [Listing Error] Mattacloud Beta

jsdokos, 2021-10-31:
### Which Plugin?

https://plugins.octoprint.org/plugins/mattacloud/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

All links go to what appears to be the old website, see below for examples:
https://mattalabs.com/products/mattacloud/
https://cloud.mattalabs.com/accounts/signup/
https://cloud.mattalabs.com/accounts/login/
https://cloud.mattalabs.com/printer-dashboard/

Link to license on Github does work
https://github.com/Mattalabs/OctoPrint-Mattacloud/blob/master/LICENSE

jneilliii, 2021-10-31:
@dougbrion, please update your plugin's links.

@jsdokos you may also want to post an issue on the plugin's repo: https://github.com/Mattalabs/OctoPrint-Mattacloud

jsdokos, 2021-10-31:
Issue has been cross posted: https://github.com/Mattalabs/OctoPrint-Mattacloud/issues/19

dougbrion, 2021-10-31:
Thanks for pointing this out, will update them this week!

cp2004, 2022-02-23:
@dougbrion are you going to sort this?

dougbrion, 2022-02-23:
Sorry it has taken so long... PhD has been getting in the way. Just submitted a pull request ( #1017 ) to fix the broken URLs.

cp2004, 2022-02-26:
PR was merged