OctoPrint/plugins.octoprint.org #961: [Listing Error] GcodeSuperLaserController

MichielO, 2021-11-14:
### Which Plugin?

https://plugins.octoprint.org/plugins/gslc/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

Broken link. The GitHub project is not found.
Is this the right project? https://github.com/jneilliii/GCodeSuperLaserController

cp2004, 2021-11-14:
Looks like the author may have deleted the repository. The link you have found is a fork of the original repository, not the actual project itself.

jneilliii, 2021-11-14:
Yeah, this looks more like an abandonment request. Can't remember why I had created a fork of that plugin. Will remove listing from plugin repo as I have no desire to maintain this one.