OctoPrint/plugins.octoprint.org #966: Adoption of Continuous Print Queue Plugin

smartin015, 2021-11-22:
### Which Plugin?

https://plugins.octoprint.org/plugins/continuousprint/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/Zinc-OS/continuousprint/issues/48

### Is there someone open to adopting it?

@smartin015

smartin015, 2021-11-22:
Note: the work I've been doing on this plugin is on my fork: https://github.com/smartin015/continuousprint

smartin015, 2021-12-12:
So, what're the next steps for this?

jneilliii, 2021-12-12:
Basically we need you to submit a PR to adopt the plugin now that more than 14 days have passed. You can follow what @Zinc-OS did when he adopted the plugin originally [here](https://github.com/OctoPrint/plugins.octoprint.org/pull/673/files). Just make similar changes to redirect the repo to your fork instead.

smartin015, 2021-12-29:
I got in contact with @Zinc-OS and we coordinated together to merge in my changes (https://github.com/Zinc-OS/continuousprint/pull/51)

Closing this out as they're keeping ownership at this time.

closed 2021-12-29 by smartin015