OctoPrint/plugins.octoprint.org #971: [Listing Error] Bed Level Visualizer

K20FRITZ, 2021-12-23:
### Which Plugin?

https://plugins.octoprint.org/plugins/bedlevelvisualizer/

### What kind of listing error are you reporting?

broken images on the listing

### What specifically is broken or wrong?

the visual will not show after the probing. new octo print install new octopi install python3 now 1.7. worked b4 the upgrade

cp2004, 2021-12-23:
Where is the broken image on that page?

K20FRITZ, 2021-12-23:
This is all I get.  I clicked the link it shows the box. I Uninstalled it and reinstalled as well. [image]

K20FRITZ, 2021-12-23:
[image]

cp2004, 2021-12-23:
Please report issues with specific plugins in the bug tracker of the plugin instead of here. This tracker is for reporting issues with the plugin repository, not issues with the plugins registered in it

What kinds of issues you should report here:

* missing or broken homepage, source or download URLs
* broken images
* incorrect compatibility information
* ... in short, anything related to the data on plugins.octoprint.org

What you should not report here:

* Any kinds of issues with the plugins themselves - report those to the plugin author instead!

closed 2021-12-23 by cp2004

K20FRITZ, 2021-12-23:
Sorry My fault.