OctoPrint/plugins.octoprint.org #982: [Abandoned Plugin] Custom Control Editor

trufy, 2022-01-13:
### Which Plugin?

https://plugins.octoprint.org/plugins/customControl/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/Salandora/octoprint-customControl

jneilliii, 2022-01-13:
The plugin's readme says it's being developed again, so not sure if this is actually abandoned or not. @Salandora could you please verify one way or another?

synman, 2022-02-09:
@jneilliii I've been directing people to your fork where you fixed the "delete all custom commands" on restart bug.

foosel, 2022-02-26:
@jneilliii that "actively worked on" status has been in there for almost two years now with no changes however, and I also haven't seen any activity from @Salandora on GitHub in quite a while. Which is totally fine, priorities shift, but considering that this particular plugin also has a somewhat nasty bug for the same almost two years now which has been fixed on your version but still not merged, I reached out on the plugin repo yesterday and am leaning towards declaring this one as abandoned if we don't hear anything back in terms of actual activity and a change to the bug situation in 14d. 

https://github.com/Salandora/octoprint-customControl/issues/42

As things stand now, people will install, the bug will strike somewhat later, and frustration is the result.

jneilliii, 2022-02-26:
No, it's the only one. I would be willing to adopt the plugin since it's technically not that different than event manager plugin and probably doesn't need much changes to remain functional. I'm still a firm believer that any setting outside of "plugins" should be part of Core though.

jneilliii, 2022-02-26:
I think GCODEViewer was the only one that has gone the other way from core to plugin. I can only think of Event Manager that has been added since plugin ecosystem was created, maybe pi support plugin wasn't there at the beginning but can't really remember.

cp2004, 2022-02-26:
3rd party plugins don't really end up becoming included in core as a bundled plugin, when I've looked back at history it is either new features in core being made as bundled plugins (usage tracking, error tracking, printer dialogs/notifications, etc.) or core components being extracted into plugins. I'm not aware of any time where it's happened before.

synman, 2022-02-26:
I guess it boils down to how important are maintaining custom controls to the functionality of the overall ecosystem.   There is some decent core functionality being exposed already to make this capability available.   I've used it within my own plugin to maintain custom controls functionality similar to what is exposed in the core control tab.

It's some pretty cool stuff.  I was surprised how easy it was to pull it all together given what the framework provides.

The only missing piece IMHO is a sure fire way to manage them in a WYSIWYG manner.

foosel, 2022-02-26:
I'm willing to make it a core plugin.

Salandora, 2022-05-20:
@foosel sorry for the late replay, go for it. 
I will probably not continue on any of my plugins for the time being. 
I don't really have time to print nowadays.

Feel free to open a Abandoned for the rest of my plugins too or if you want put them as a core plugin.

foosel, 2022-06-21:
See OctoPrint/OctoPrint#4526

foosel, 2026-09-04:
For the record, this plugin's functionality has been made available through the bundled Custom Control Manager plugin since OctoPrint 1.11.0.

I've also just marked it as abandoned (which we so far forgot to do for some reason), added a notice about it and put it on the blocklist for OctoPrint 2.0.0+.

Long term we might want to through it out of the repo altogether but for now I think that's enough.

closed 2026-09-04 by foosel