OctoPrint/plugins.octoprint.org #983: [Abandoned Plugin] Octoprint-Smart-Filament-Sensor

drbrown-l3tech, 2022-01-16:
### Which Plugin?

https://plugins.octoprint.org/plugins/smartfilamentsensor/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/maocypher/Octoprint-Smart-Filament-Sensor

drbrown-l3tech, 2022-01-16:
The author has posted that it is abandoned on the bottom of the Readme.

jneilliii, 2022-01-27:
@Royrdan, few things I'll need from you. 

1. Enable issues on your fork of the repository.
2. Update the user [here](https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor/blob/ce247f3f4ebb3e72e65dfe9ea95b9659ee1db3f2/octoprint_smart_filament_sensor/__init__.py#L348)
2. Update the url [here](https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor/blob/ce247f3f4ebb3e72e65dfe9ea95b9659ee1db3f2/octoprint_smart_filament_sensor/__init__.py#L369).
3. Bump the version [here](https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor/blob/ce247f3f4ebb3e72e65dfe9ea95b9659ee1db3f2/setup.py#L18) to `1.1.6`.
4. Create a new tag named `1.1.6` and release with the above changes included.

jneilliii, 2022-01-29:
@Royrdan I see you took care of 1-4, small correction needed. GitHub is case sensitive on the URL so make the line [here](https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor/blob/bffb42fa6350539a523ee4db132235b5a56ad3c3/octoprint_smart_filament_sensor/__init__.py#L369) be `pip="https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor/archive/{target_version}.zip"`, notice capital `R` in username. Also Go [here](https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor/releases) click new release button in the upper right and enter 1.1.6 for the tag, give it a name and add any release notes you'd like to include about adopting the plugin. That will handle `#5`. Please reply here once complete so I know it's done.

Royrdan, 2022-01-30:
Hi jneilliii,
Thanks for your help with that.
It should all be completed now.
Let me know anything else.
Thanks

closed 2022-01-31 by jneilliii

joshuacant, 2022-01-31:
Hi, I am unsure if this is the right place for this, please accept my apologies in advance if it is not!

I got an update notice on this plugin today and when I updated it, the following errors were produced:

> Updating, please wait.
> +++++++++++++++++++++++++++++++++++++++++++
> Now updating Smart Filament Sensor to 1.1.6
> +++++++++++++++++++++++++++++++++++++++++++
> /home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.6.zip --no-cache-dir
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.6.zip
> ERROR: HTTP error 404 while getting https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.6.zip
> ERROR: Could not install requirement https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.6.zip because of HTTP error 404 Client Error: Not Found for url: https://codeload.github.com/maocypher/Octoprint-Smart-Filament-Sensor/zip/1.1.6 for URL https://github.com/maocypher/Octoprint-Smart-Filament-Sensor/archive/1.1.6.zip
> The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

It seems like the updater is confused and trying to pull the new version from the previous repo. Is there any way to fix this?

cp2004, 2022-01-31:
Potential fix in #1002

joshuacant, 2022-02-01:
```
Updating, please wait.
+++++++++++++++++++++++++++++++++++++++++++
Now updating Smart Filament Sensor to 1.1.6
+++++++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor/archive/1.1.6.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor/archive/1.1.6.zip
Downloading https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor/archive/1.1.6.zip
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-a_aihw9q/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-a_aihw9q/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-kgqkqz5k
cwd: /tmp/pip-req-build-a_aihw9q/
Complete output (6 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-req-build-a_aihw9q/setup.py", line 25
plugin_author = Royrdan"
^
SyntaxError: EOL while scanning string literal
----------------------------------------
WARNING: Discarding https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor/archive/1.1.6.zip. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.
```

Looks like a missing double quote, but I see the setup.py in your repo was fixed. Maybe you need to push a new release zip with that change?

jneilliii, 2022-02-01:
It does look like you are correct @joshuacant. The tagged 1.1.6 version is missing the first double quote for author. @Royrdan you need to re-create that tag and force push or bump the version in setup.py and create a new 1.1.7 version tag and release.

JodyMT, 2022-03-04:
@Royrdan - if you are still maintaining this plugin THANK YOU 1st of all, is it possible to add some sort of distance calibration to the plugin (i made a custom filament sensor and i'm sure the distance calibration is off)

cp2004, 2022-03-04:
@JodyMT Feature requests should go directly to the plugin author, not here.

https://github.com/Royrdan/Octoprint-Smart-Filament-Sensor

Royrdan, 2022-03-04:
> @Royrdan - if you are still maintaining this plugin THANK YOU 1st of all, is it possible to add some sort of distance calibration to the plugin (i made a custom filament sensor and i'm sure the distance calibration is off)
> 
> side note - the updating is still broken

Hi Jody,
Thanks for your feedback.
If you completely remove the old plugin then re-add the plugin. It should fix the issue of not updating. Please let me know if any issues with upgrading.
The plugin does not measure the filament for the moves. It works by checking if the filament has moved by the G command then if any movement is detected at all it will reset the counter. The sensor doesnt calculate how much has moved. If you are having issues with false positives I suggest you increase the detection distance which will allow the filament to move that amount and wait for any movement of the filament in the sensor before triggering. Let me know if you need more info or if I missunderstand your issue. Thanks

JodyMT, 2022-03-04:
hi, sorry about that but i can't figure out where that is, is that "pull
request" ? i dont know how to contact him

Royrdan, 2022-03-04:
@JodyMT See previous reply I replied to you. But also if you have an issue you can raise it under the link in the provided link above under the issues tab and create a new issue. But please read the above the reply I have posted if that answers your questions. Thanks

PR #998 (2022-01-31): smart filament sensor adoption
adoption of smart filament sensor plugin, resolves #983 once requested changes are completed on forked repo. need to update authors list with how @Royrdan wants to be listed.