OctoPrint/plugins.octoprint.org #994: [Abandoned Plugin] Filament Sensor Simplified

Joost79, 2022-01-25:
### Which Plugin?

https://plugins.octoprint.org/plugins/filamentsensorsimplified/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/LuckyX182/Filament_sensor_simplified/issues/38

Joost79, 2022-01-25:
The last reply from the developer was in September, no reply/activity since, jet there were users asking for feedback.

jneilliii, 2022-01-25:
Just because there's no activity in a few months doesn't mean it's abandoned by @LuckyX182, but they could be a student and back in school or something similar. That being said, there is an open PR and several unanswered issues on the repo indicating potential abandonment.

jneilliii, 2022-01-25:
If @LuckyX182 doesn't respond here within 14 days I will mark plugin abandoned.

LuckyX182, 2022-01-31:
Plugin not abandoned but don't have much time for it.

cp2004, 2022-01-31:
Thanks for getting back about the plugin's status, this issue can be closed for now then.

closed 2022-01-31 by cp2004