OctoPrint/plugins.octoprint.org #1006: [Listing Error] Fractal Bridge

andyhud09, 2022-02-09:
### Which Plugin?

https://plugins.octoprint.org/plugins/fractalbridge/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

The Discord invite has expired.

foosel, 2022-02-09:
Hi @Popeyef5, see above report. Also seems to apply to the invite in your repository's README.

foosel, 2022-02-25:
@Popeyef5 ping?

foosel, 2022-02-25:
@FractalManufacturing @Fractal3D see above

Popeyef5, 2022-03-09:
Hey! Nice catch. Sorry for the late reply, I was AFK for the past few weeks. I'll fix it ASAP. Thanks again!

jneilliii, 2022-04-13:
Any update @Popeyef5 ?

foosel, 2024-06-12:
Two years of silence and the website no longer works either. I guess it's time to delist this.

commit 92b47d47 (2024-06-12): Unlist Fractal Bridge

Closes #1006
closed 2024-06-12 by foosel