OctoPrint/plugins.octoprint.org #1010: [Listing Error] OctoPrint-TuyaSmartplug

timbarnes, 2022-02-12:
### Which Plugin?

https://plugins.octoprint.org/plugins/tuyasmartplug/

### What kind of listing error are you reporting?

incorrect compatibility information on the plugin listing

### What specifically is broken or wrong?

The image showing the interface is no longer correct. There is an additional button for "Version 3.3" in the current interface, which does not appear to be documented.

cp2004, 2022-02-23:
It is difficult for plugin authors to keep the plugin repository up to date with exactly how the plugin looks when it is installed, the plugin repository listing is there to give you an idea of what the plugin can do, rather than a complete guide to the plugin. If that's necessary to use it, it should be documented separately.

@ziirish are you able to sort this?

ziirish, 2022-02-24:
Hello, I'll make new screenhots when I have a chance.
Thanks for the report!