OctoPrint/plugins.octoprint.org #1018: [Abandoned Plugin] OctoPrint-Enclosure

foosel, 2022-02-25:
### Which Plugin?

https://plugins.octoprint.org/plugins/enclosure/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/435

foosel, 2022-02-25:
Timeline: 

  * 2021/08/30 - originally inquiry by @hbast if plugin was still actively maintained
  * 2021/10/12 - I asked about status after reported incompatibilities with latest OctoPrint release. @vitormhenrique gets back to me, confirms they are still working on it but need help, I give some exposure on Twitter. A ton of people chime up in the thread, a Discord server is created
  * 2021/10/18 - Latest release
  * 2022/02/22 - [worrying report from a frustrated @hbast](https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/435#issuecomment-1047943390), ticket closed by them

Current state: 149 open issues and 5 PRs

It feels like this is a popular plugin with a maintainer who can't give it the time it needs right now. Which is totally fine, happens, I completely get it. It should be made more transparent however as there might be people open to taking over maintenance who can give it more of the attention it apparently requires.

rtheil-growlink, 2022-02-25:
If I didn't loathe Python, I would absolutely contribute. I do love this plugin and am not aware of an equivalent alternative. I'm concerned that I'll lose use of the plugin if I upgrade Octoprint, especially with the upgrade from Python 2.x to 3. Another reason why I dislike Python.

cp2004, 2022-02-25:
The plugin says it's compatible with Python 2 and 3. This is not the place for Python bashing please, constructive comments would be preferred.

rtheil-growlink, 2022-02-25:
> The plugin says it's compatible with Python 2 and 3. This is not the place for Python bashing please, constructive comments would be appreciated.

Except that it's not entirely compatible. Several open issues related to the plugin having errors after upgrading to Python 3. This is not meant to bash Python, just to state the fact that what works with Python 2 won't always work with Python 3, and those issues are extremely common. It is also a reason why some developers can't/won't contribute. Vitor even stated on Oct 12 that the project needed to "Migrate to the new branch on python3".

vitormhenrique, 2022-02-26:
i think there is a misunderstanding here.

I said that I was gonna make this plugin compatible with python 3, and we got those pull requests on.

I'll accept PR's for but fixes but NOT new features.

The current way that the plugin works takes a lot of effort and knowledge of python environments and libraries to make it work, the configuration is terrible and user's just don't know how to use it.

The idea here is to move away from this plugin and build another with helper plugins, each helper plugin would have all libraries and easier way to configure them.
For example a GPIO helper plugin would be able to act like a trigger and actions.
The main plugin would have the glue logic and enable a screen like an "IF THIS THEN THAT".

The POC with the new style of plugin was here on the "Python 3" branch. But moved it to here: 
https://github.com/vitormhenrique/OctoPrint-Enclosure-V2

An example of a helper plugin is here:
https://github.com/vitormhenrique/OctoPrint-Enclosure-RPiGPIO-Faker

This is only a POC and I already have some stuff that I don't like and will be changing. 

I still don't know how the temperature control would be implemented. I like the idea to build something like an if this then that, but I don't think it fits well with a trigger -> action based scheme (or maybe it does? if temperature read 60C -> action control temperature? But the controlling of temperature would depend on another action, meaning, turn on / off a GPIO, send a message to a MQTT broker etc).

The idea would be something like the following:

[image]

foosel, 2022-02-26:
@vitormhenrique That does sound however like v1 (which currently the only one available) is pretty much abandoned since you are working on v2? Or will you also still push out new versions if there are incompatibilities or anything like that? I'm only trying to understand the status here and make sure it is correctly reflected to the community in the plugin repository. From the feedback on the ticket on your repo (which I got repeatedly dragged back into due to mentions) it sounded like there was some stalling, but I might have misunderstood the situation. If the plugin in its current iteration is feature complete as it is right now according to your plans, but you are still on standby for it in case there are bugs that cause issues (as in, happy to merge or just create fixes yourself *and* release a new version on time), then we can happily close this again.

vitormhenrique, 2022-02-26:
I plan to keep it compatible with no more features added.
The issue with the current plugin is that it requires a lot of knowledge, a lot of people can't get the plugin working because they don't know how to install and properly configure.
I'm happy to merge PR's with fixes and I have done so 5 times this year.
I'm also happy to add anyone as collaborators, if they want to improve or add other stuff.
But I haven't got anyone requesting to be a collaborator yet.

foosel, 2022-02-26:
Then this sounds more like a communication issue and misunderstanding and I apologize.

closed 2022-02-26 by foosel

rtheil-growlink, 2022-02-27:
Thanks for the update, @vitormhenrique. Good to know the plugin has been updated to work with Python 3. I feel safe upgrading octoprint and Python to latest now.