OctoPrint/plugins.octoprint.org #1021: [Abandoned Plugin] Continuous Print

willindiana, 2022-02-28:
### Which Plugin?

https://plugins.octoprint.org/plugins/continuousprint/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/Zinc-OS/continuousprint/issues/70

### Is there someone open to adopting it?

Primary: @smartin015  Fallback: @willindiana

smartin015, 2022-02-28:
Some context: I'd previously opened #966, closing after I heard back from @Zinc-OS. But nobody's heard from them in over a month now, and contributions are piling up. I'm still open to adopting this plugin.

willindiana, 2022-02-28:
I second that sentiment as I have had an open PR since October of last year. https://github.com/Zinc-OS/continuousprint/pull/46. I have been watching Scott respond and patch issues diligently and I trust he is capable of taking over.

jneilliii, 2022-02-28:
@Zinc-OS I totally understand life happens and other things take priority over plugin development, but it seems you have a couple of people that are willing to help. Would you be up for adding these guys as contributors to your repo or should we consider this abandoned and transfer ownership?

arcasoy, 2022-03-12:
@jneilliii May I ask what are the next steps in the process? Hoping to see progress soon from those able to contribute.

jneilliii, 2022-03-13:
Since it appears Zinc-OS has gone the next steps are to create a fork and merge any outstanding PRs/bug fixes and release as a new version on your fork, making sure to update the version in setup.py. Once that is completed submit a PR here adding yourself to the authors list in the template and update the links and such to your fork. We'll then review that PR and verify everything is setup properly to redirect people to your new fork for the new version.

jneilliii, 2022-03-13:
Here's an example of an adoption PR...

https://github.com/OctoPrint/plugins.octoprint.org/pull/998/files

PR #1027 (2022-03-13): Adopt continuousprint plugin
Fixes #1021

I completed https://github.com/smartin015/continuousprint/pull/5 to bump the plugin version and replace occurrences of the old repository. Then I created a new release: https://github.com/smartin015/continuousprint/releases/tag/1.4.1

The changes in the new release have been simmering for over a month (see https://github.com/smartin015/continuousprint/issues/3) so odds are good of a (relatively) painless switchover.  I *think* I got all the references, but I'd really appreciate some careful review to make sure.

Let me know if there's anything extra I  [... cut]
closed 2022-03-13 by jneilliii