OctoPrint/plugins.octoprint.org #1035: [Abandoned Plugin] OctoPrint-Prusa-Mini-ETA

sarusani, 2022-03-22:
### Which Plugin?

https://plugins.octoprint.org/plugins/prusa_mini_eta/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/vookimedlo/OctoPrint-Prusa-Mini-ETA/issues/6

jneilliii, 2022-03-22:
It appears that an open PR on the original repo may resolve the issue linked [here](https://github.com/egguy/OctoPrint-Prusa-Mini-ETA). Maybe @egguy would be up for adoption?

egguy, 2022-03-22:
I would be OK to adopt the plugin. Should I do a fork of the repo and change [the .md ](https://github.com/OctoPrint/plugins.octoprint.org/blob/gh-pages/_plugins/prusa_mini_eta.md) ?

cp2004, 2022-03-22:
Is there a difference between the aforementioned plugin and this one? https://plugins.octoprint.org/plugins/m73etaoverride/

SB97, 2022-03-22:
@cp2004 just looked at the source code of both. They have the same functionality. And It looks like neither of those do support the silent mode in PrusaSlicer.

egguy, 2022-03-22:
There's an unmerged PR to have the estimated time from the initial gcode loading, which is already present in this plugin.
@SB97 If there's a way to detect if the printer is in silent mode, I could try to implement. (there's also a separate branch with this functionality, but unfinished)

egguy, 2022-03-22:
Ok, I've implemented a support for the silent mode of the Prusa MK3, but it requires activating the functionality in the plugin's settings.
I think the plugin should work with other printers, but I can't test :s

jneilliii, 2022-03-23:
> I would be OK to adopt the plugin. Should I do a fork of the repo and change [the .md ](https://github.com/OctoPrint/plugins.octoprint.org/blob/gh-pages/_plugins/prusa_mini_eta.md) ?

The process is fairly easy. You just submit a PR adding you to the authors list in the md file, and preferably the overlays for the update checks. You'll want to finalize any changes and bump the version in setup.py and create a release in GitHub with the same version as the tag. Here's an example of a recent adoption: https://github.com/OctoPrint/plugins.octoprint.org/pull/1027/files

jneilliii, 2022-03-23:
but agree with @cp2004 if the other plugin does the same thing and is still active it would probably be better to retire one and keep the other. There may have been a technical difference to how the plugins were functioning, but further review would need to be made for that.

jneilliii, 2022-03-23:
I remember now why there are two. Prusa-Mini-ETA also includes an analysis factory, whereas M73ETAOverride is just an estimation factory.

closed 2022-04-01 by jneilliii