OctoPrint/plugins.octoprint.org #1044: [Abandoned Plugin] FileManager

wizard04wsu, 2022-04-10:
### Which Plugin?

https://plugins.octoprint.org/plugins/filemanager/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/Salandora/OctoPrint-FileManager#readme

cp2004, 2022-04-10:
Not the first time this question has been asked. See #453 #719 & https://github.com/Salandora/OctoPrint-FileManager/issues/35

wizard04wsu, 2022-04-10:
Well, it's probably time to ask it again. This plugin has 13 outstanding issues and there have been no changes to the repository and no comments made by the owner in over a year.

foosel, 2026-09-04:
For the record, this plugin's functionality has been made available through the bundled Upload Manager plugin since OctoPrint 1.11.0.

I've also just marked it as abandoned (which we so far forgot to do for some reason), added a notice about it and put it on the blocklist for OctoPrint 2.0.0+.

Long term we might want to through it out of the repo altogether but for now I think that's enough.

closed 2026-09-04 by foosel