OctoPrint/plugins.octoprint.org #1048: Deregistration of CrealityCloud plugin due to copyright situation

foosel, 2022-04-14:
It has come to our attention that there are allegations of content theft from prominent members of the community against your platform, as of yet unaddressed by you:

  * https://twitter.com/agepbiz/status/1514611716985790475

Due to this, we are removing this listing from the plugin repository again. Once these complaints have been resolved we will reconsider adopting the plugin into the official repository by reverting the above commit.

_Originally posted by @foosel in https://github.com/OctoPrint/plugins.octoprint.org/issues/959#issuecomment-1099308355_

@crealitycloud Please investigate and resolve this with the content author in question. As said in #959 we are happy to reregister the plugin here once that has been done.

crealitycloud, 2022-04-21:
Hello,we have been in contact with Stian Ervik Wahlvåg and removed all the models that violated his copyright. We are a platform that provide service not content. We did not upload these models and are not involved in any of these copyright infringement behaviors. We vigorously protect intellectual property rights and take reports of suspected infringements seriously. To be clearer, it was not us that upload these models and we check reports in a daily basis to remove any inappropriate content.

crealitycloud, 2022-04-22:
Protecting the rights of creators has always been at the core of our operations. Here is the list of models that we have verified and removed. The list will be continually updated.
https://github.com/crealitycloud/deleted_models/blob/main/README.md

closed 2022-06-21 by jneilliii