OctoPrint/plugins.octoprint.org #1060: [Listing Error] ExtraFileInfo

theturtle32, 2022-05-06:
### Which Plugin?

https://plugins.octoprint.org/plugins/extrafileinfo/

### What kind of listing error are you reporting?

incorrect compatibility information on the plugin listing

### What specifically is broken or wrong?

This plugin depends on a separate OctoPrint plugin [OctoPrint-SlicerSettingsParser-Python3](https://github.com/Rob4226/OctoPrint-SlicerSettingsParser-Python3) already being installed, but does not specify that you must manually install the dependency. It links to the plugin's GitHub repo, saying that it uses it, but doesn't indicate that you must install that library/plugin separately using manual installation before you can install this plugin. OctoPrint-SlicerSettingsParser-Python3 does not appear when searching for plugins in the plugin manager and must be installed via its package URL instead.

jneilliii, 2022-05-06:
The plugin author is working on updating the plugin listing with more detailed instructions.

larsjuhw, 2022-05-09:
[Should be fixed now.](https://github.com/larsjuhw/OctoPrint-ExtraFileInfo/commit/381fba3a4b2fe04020e2fbc11a6384edf59a567f#diff-60f61ab7a8d1910d86d9fda2261620314edcae5894d5aaa236b821c7256badd7R36)

closed 2022-05-09 by jneilliii