OctoPrint/plugins.octoprint.org #1082: [Abandoned Plugin] OctoPrint-Thingiverse

TriBird-GitHub, 2022-06-28:
### Which Plugin?

https://plugins.octoprint.org/plugins/thingiverse/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/stefancandrea/OctoPrint-Thingiverse/blob/master/README.md

### Is there someone open to adopting it?

I tried but then realized it's kind of a security risk and just skips 1 step

ender-bit, 2025-03-05:
ill take over if you want me to

jacopotediosi, 2025-05-20:
@foosel @stefancandrea 

I feel that this plugin should be removed and/or blacklisted, as it appears to be very dangerous.

At some point after the plugin’s initial release, the Thingiverse website introduced security policies to prevent the site from being embedded in iframes. Since this change made the plugin unusable, the author updated the plugin listing on [plugins.octoprint.org](https://plugins.octoprint.org/plugins/thingiverse/), as well as the settings tab and README, suggesting users to install third-party browser extensions and configure them to disable fundamental browser security protections such as the [Content Security Policy](https://developer.mozilla.org/en-US/docs/Web/HTTP/Guides/CSP).

These configuration changes are dangerous and — if followed as instructed by the plugin author — are applied globally, meaning that users will be disabling these protections not only for OctoPrint but for all websites they visit.

[image]

The issue with this plugin is at its root. It needs to be completely rewritten using the Thingiverse API.

Attempting to embed the Thingiverse website in an iframe, or trying to bypass the `X-Frame-Options` security protection — as seen in [this commit](https://github.com/stefancandrea/OctoPrint-Thingiverse/commit/3a741975b2e6fcc5898bc5e7cd6c6b4edce64b1b) — are not viable solutions given how modern browsers work and the current standards for web security.

foosel, 2025-05-20:
Agreed. I've just removed it (this time also including the assets I hope)

commit faca5b2e (2025-05-20): chore: remove thingiverse plugin

Closes #1082
closed 2025-05-20 by foosel