OctoPrint/plugins.octoprint.org #1085: [Abandoned Plugin] Flashforge/Dremel/PowerSpec Printer Support

Z-K-O, 2022-07-01:
### Which Plugin?

https://plugins.octoprint.org/plugins/FlashForge/

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/Mrnt/OctoPrint-FlashForge/blob/master/README.md

Z-K-O, 2022-07-01:
Copied directly from readme:

> **March 9th 2022 - IMPORTANT PLEASE READ**
Neither FlashForge or Dremel have provided details of the commands or protocol supported by the various models - almost all the work herein was done by reverse engineering the communication between FlashPrint and the two printers I had access too, trial and error, and subsequently spending many, many hours studying debug logs including those graciously provided by users.

> However, this is ultimately a losing battle - every new device or firmware change by FlashForge can and has broken the limited capabilities of this plugin. I have therefore made the decision not to spend more time on this plugin. I am grateful to the few users that were generous enough to send me a donation and I hope that they will understand this decision. Moving forward, the best solution for FlashForge printers is to use a well documented, open source firmware such as Marlin, which will fully support OctoPrint, Cura, etc. I am working on converting my remaining FlashForge printer to use Marlin firmware so that I can focus on using it to its full potential.

chessdenman, 2022-08-19:
Hi,
I'd be very interested in helping out and learning about the merlin convert of the printer.
I have not got the skills to do reverse engineering on my creator pro and have simply frozen the version of flashprint I use and my version of octoprint so as to keep using your plugin.
My guess it that flashforge will simply abandon the printer ultimately as it is aging (in not being directly connected to the internet).
Are you giving your printer a new control unit?

cp2004, 2022-08-19:
@chessdenman - this ticket relating to the listing of the plugin in OctoPrint's plugin repository.

If you would like to discuss connecting/using FlashForge printers or the development of this plugin, please use the community forums or discord server, as well as any resources on the actual plugin's repository.

Commander-Firestrike, 2022-09-17:
I can't find @chessdenman on Discord or the Octoprint Forums so I'm messaging here.

I managed to convert my Inventor to Marlin with the [moonglow/Flashforge_Marlin](https://github.com/moonglow/FlashForge_Marlin) repo, maybe that would help you? Feel free to contact me on [my profile on the Octoprint forums](https://en.wikipedia.org/wiki/Inherently_funny_word).

Mrnt, 2022-09-25:
@chessdenman I doubt FlashForge will abandon their printer line - they keep on making more with seemingly the same closed source code base. They do however stop supporting old versions of printer that they branded for someone else, eg the PowerSpec Ultra for MicroCenter. I suggest you look into the [moon glow/Flash](https://github.com/moonglow/FlashForge_Marlin) project as suggested by @Commander-Firestrike. This is a bit fiddly to install on the first pass and you may lose WiFi access but you will gain Marlin compatibility along with the ability to use other open source tools such as Ultimaker Cura as a slicer.

msmthng, 2023-02-11:
Did sone updates in a fork, forked version is running with 1.8.x again. Merge request started.