OctoPrint/plugins.octoprint.org #1097: [Listing Error] OctoEverywhere - Free Remote Monitoring, Notifications, And Control!

martin-nieuwoudt, 2022-08-01:
### Which Plugin?

https://plugins.octoprint.org/plugins/octoeverywhere/

### What kind of listing error are you reporting?

incorrect compatibility information on the plugin listing

### What specifically is broken or wrong?

You are reporting octoeverywhere as free. It is not. This is false advertising. Stop lying to the public

jneilliii, 2022-08-01:
@QuinnDamerell has your policy changed with octoeverywhere?

QuinnDamerell, 2022-08-01:
@jneilliii nope, everything is still the same as it's always been!

@martin-nieuwoudt, I'm sorry the plugin seemed incorrect to you. I'm happy to help you figure out the discrepancy either in this thread, or you can email me privately using: https://octoeverywhere.com/support

My goal from the beginning of the project has always been to make the service as free as possible, so everyone in the OctoPrint community can take advantage of it. As features have been added, I have tried to keep them all also as free as possible, but I do have to gate something behind my supporter model because they cost more to operate. For example, I allow for five email notifications a day for all users and unlimited for supporters. Each email costs my service money to send, so I have to limit emails for those who aren't supporting. 

Anyways, I'm happy to hear more about your concern and help you out!

closed 2022-08-01 by jneilliii