OctoPrint/plugins.octoprint.org #1101: [Listing Error] Prusa MMU plugin download link is broken (2022.08.14)

nikkolade, 2022-08-17:
### Which Plugin?

https://plugins.octoprint.org/plugins/prusammu/

### What kind of listing error are you reporting?

broken link on the listing

### What specifically is broken or wrong?

The latest release (2022.08.14) download is broken.
The broken link is:
https://github.com/jukebox42/Octoprint-PrusaMMU/releases/2022.8.14/download/Octoprint-PrusaMmu.zip
where the date and download directories are swapped.

The working link is:
https://github.com/jukebox42/Octoprint-PrusaMMU/releases/download/2022.8.14/Octoprint-PrusaMmu.zip

jneilliii, 2022-08-17:
This is actually an error with the plugin's software update hook configuration, not the plugin listing itself. You should report to that plugin's repository if you haven't already.

jneilliii, 2022-08-17:
I went ahead and posted an issue, linked above.

closed 2022-08-17 by jneilliii