OctoPrint/plugins.octoprint.org #1118: Repository will not update Github information correctly

NilsRo, 2022-10-28:
As this seems not a content error I opened a blank issue. 

Discussed here https://community.octoprint.org/t/plugin-repos-github-stats-shows-old-release-as-most-actual/47467 the Github information of plugin https://plugins.octoprint.org/plugins/SlicerEstimator/ is not updated correctly. It has shown a more actual release in the past but switched back to 2021 actually. But the count of releases is fine. You can see the difference in the screenshots.

[image]
[image]

foosel, 2022-10-31:
I don't know yet why, but for some reason when I query the last release on GitHub's GraphQL API, this is what I get back:

``` json
{
  "data": {
    "repository": {
      "lastRelease": {
        "nodes": [
          {
            "name": "erstes SlicerEstimator Release",
            "publishedAt": "2021-02-22T00:06:11Z",
            "url": "https://github.com/NilsRo/OctoPrint-SlicerEstimator/releases/tag/0.4",
            "tagName": "0.4",
            "isPrerelease": false
          }
        ]
      }
    }
  }
}
```

foosel, 2022-10-31:
Something about the default sorting order of the `releases` query changed.

```
query {
  repository(owner: "NilsRo", name: "OctoPrint-SlicerEstimator") {
    lastRelease: releases(last:1){
       nodes {
        name,
        publishedAt,
        url,
        tagName,
        isPrerelease
      }
    },
  }
}
```

This used to give me the latest release. Now it gives me first. So apparently the default sorting direction changed. I'll change the query to explicitly state the sorting order and field.

```
query {
  repository(owner: "NilsRo", name: "OctoPrint-SlicerEstimator") {
    lastRelease: releases(first:1, orderBy:{direction:DESC, field:CREATED_AT}){
       nodes {
        name,
        publishedAt,
        url,
        tagName,
        isPrerelease
      }
    },
  }
}
```

closed 2022-10-31 by foosel