OctoPrint/plugins.octoprint.org #1122: [Abandoned Plugin] Exclude Region

katiekloss, 2022-11-13:
### Which Plugin?

https://plugins.octoprint.org/plugins/excluderegion

### Link to ticket on plugin's issue tracker confirming abandonment

https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/pull/64

### Is there someone open to adopting it?

@katiekloss

jneilliii, 2022-11-13:
Thanks @katiekloss for volunteering. Process is pretty straight forward. Fork the plugin to your account, make the changes to bring it up to date and update setup.py with incremented version and create a matching tag/release on your repo. Then check other recent adoption PRs for what kind of changes are required here. Happy to answer any questions.

cp2004, 2022-11-13:
> Link to ticket on plugin's issue tracker confirming abandonment
> https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/pull/64

I think someone should probably open an issue on the repository to at least make it more explicit that the plugin is abandoned and could be adopted. Sometimes when asked directly people will respond and may transfer ownership for example, making the transition basically seamless. Rather than linking to an old PR, that makes no mention of abandoned status.

katiekloss, 2022-11-14:
> I think someone should probably open an issue on the repository to at least make it more explicit that the plugin is abandoned and could be adopted. Sometimes when asked directly people will respond [...]

Good point, updated the original issue with a new one I made in that repo.

chunkysteveo, 2022-12-02:
It's a beautiful plugin that's got me out of so many holes when printing large plates of parts! I hope someone can pick this up 🙏

pneumaticdeath, 2022-12-02:
I'm pretty handy with python on the backend, and would love to help.  Please let me know if I can be of assistance.

WesMcGee68, 2022-12-03:
I have it installed but I'm not too savvy with the plugin and haven't been able to get it set up properly.   Would it be possible to put a walk through on it's setup for the beginners, on the home page??

closed 2022-12-04 by jneilliii

bradcfisher, 2022-12-07:
I have not abandoned the plugin, though I have also not had as much free time to maintain it as I'd like lately either.  I do plan to continue working on it as time permits, and have several enhancements planned.

I certainly don't mind @katiekloss creating a fork and releasing a derivative, as that's the open source way and permitted by the licensing.  There's nothing wrong with giving folks choices.

bradcfisher, 2022-12-07:
To follow up on my previous comment, I would suggest that @katiekloss should instead release a separate plugin under a different name and versioning to avoid confusion.  Proper attribution should also be maintained.

jneilliii, 2022-12-07:
@bradcfisher, if time is an issue maybe you guys could join forces and you could add @katiekloss to your repo as a maintainer.

PR #1132 (2022-12-04): Update authors, home links to adopt ExcludeRegion
Resolves #1122